dmitrystu/libusb_stm32 #1: Interrupt Endpoint callbacks missing

xcvista, 2017-10-20:
I am trying to build a HID device, which calls for a pair of Interrupt endpoints. However the endpoint callbacks are not being called at all, although on the host side the system is trying to INTERRUPT IN.

What is amiss here? Is there something I need to be aware of?

Here is [my code](https://github.com/SushiBits/LightSwitchUSB-Firmware/blob/master/src/usb_device.c). I am using an interrupt-driven stack so this file is pretty self-contained.

dmitrystu, 2017-10-21:
Well. Endpoint TX event raises after TX completion. Therefore you should transmit 1'st packet asynchronously. Try to send zero-length packet to IN endpoint just after endpoint configuration.

```c
        usbd_ep_config(dev, USB_RXD_EP, USB_EPTYPE_INTERRUPT, USB_PKT_SZ);
        usbd_ep_config(dev, USB_TXD_EP, USB_EPTYPE_INTERRUPT, USB_PKT_SZ);
        usbd_reg_endpoint(dev, USB_RXD_EP, usb_rx);
        usbd_reg_endpoint(dev, USB_TXD_EP, usb_tx);
        usbd_ep_write(dev, USB_TXD_EP, 0, 0);
        return usbd_ack;
```

xcvista, 2017-10-21:
@dmitrystu Thanks. I added the write and it worked. Another problem is about reenumeration. Why it isn't quite working?

dmitrystu, 2017-10-21:
At this moment i have no idea what's happens. I will try your code ASAP. Could you explain me how i can reproduce this problem?

xcvista, 2017-10-21:
Feed it a bad descriptor or refuse it the HID Report Descriptor, and Windows will start reenumerating.

dmitrystu, 2017-10-24:
Found bug in OTG_FS driver that freeze MCU on USB bus reset when endpoints is configured (reenumeration), No similar problems found for other drivers. See "dev" branch.

closed 2018-01-03 by dmitrystu