dmitrystu/libusb_stm32 #5: Missing stm32.h

AlbanBedel, 2017-11-01:
I'm trying to compile this library on Ubuntu for an STM32F048xx. I downloaded STM32CubeF0 v1.9.0 and added the following to the Makefile:
```
STARTUP.stm32f048xx  = $(CMSISDEV)/STM32L0xx/Source/Templates/gcc/startup_stm32f048xx.s
CFLAGS.stm32f048xx   = -mcpu=cortex-m0
DEFINES.stm32f048xx  = STM32F0 STM32F048xx
```

And try to build the module, but it fails:
```
$ make module CMSIS=../STM32Cube_FW_F0_V1.9.0/Drivers/CMSIS MCU=stm32f048xx
compiling src/usb_32v1.o
src/usb_32v1.c:18:19: fatal error: stm32.h: No such file or directory
 #include "stm32.h"
```

I'm new to the STM32 controllers so perhaps that should be obvious, but I wasn't able to find which library I'm missing here.

dmitrystu, 2017-11-01:
Use this [stm32.h](https://github.com/dmitrystu/stm32h)
Put it into ../STM32Cube_FW_F0_V1.9.0/Drivers/CMSIS/Device/ST/

AlbanBedel, 2017-11-02:
Thanks! Perhaps this should be mentioned in the readme.

closed 2017-11-02 by dmitrystu