dmitrystu/libusb_stm32 #8: Interrupt example?

kisielk, 2017-11-29:
First of all, thanks for writing this library. I'm seriously considering using it over the official ST libs which are a pretty big mess when it comes to USB especially.

Do you have a code example of the library with interrupts? The demo seems to use polling, unless I'm missing something...

dmitrystu, 2017-11-29:
It's easy. Enable USB IRQ and use usbd_poll(&udev) in the IRQ handler.
```
void USB_IRQHandler(void) {
    usbd_poll(&udev);
}
```
It worked at least for L052 and L100. 
I'll add a code example uses interrupts to CDC loop demo.

kisielk, 2017-11-29:
Simple enough. Do you happen to have a DFU example somewhere? I'm doing a project with the L052 and discovered the embedded bootloader doesn't have USB DFU, unlike most other ST chips with USB, so I'm looking at rolling my own.

dmitrystu, 2017-11-29:
About IRQ: Just checked on bluepill. It's works.
About DFU: Is [this project](https://github.com/dmitrystu/sboot_stm32) good for you ?

kisielk, 2017-11-29:
Oh yeah, that's perfect! :) Seems you've done most of the work for me! I'll give it a try.

closed 2017-11-30 by dmitrystu

xcvista, 2018-01-03:
https://github.com/SushiBits/DAP42-Firmware/blob/master/src/usb-device.c

Here is my example, with the USB started during C runtime initialization, and interrupt driven programming. My main file is simply a line of `for (;;) __WFE();`.

adadurov, 2020-01-25:
Anyone tested the interrupt mode with bluepill? Recently, it didn't work for me.

dmitrystu, 2020-01-25:
See no problems with bluepill (STM32F103C8T6) for both interrupt and non-interrupt builds.

adadurov, 2020-01-28:
Dmitrystu, many thanks for the quick response. I'm pretty sure I attempted reconnection a few times, but will give it yet another try.
By the way, can you share some insights on how you do debugging of USB code? What are the tools or approaches that you use?
I did some USB debugging a few years ago on ATMEGA by logging to a serial port and then analyzing the execution path. Do you know a better/more efficient way to do that?

xcvista, 2020-01-28:
I would suggest USB packet sniffing with Wireshark as a beginner.

dmitrystu, 2020-01-28:
@adadurov Is non-interrupt build working for you?
Since I'm under Linux I'm using the following toolset:
+ Vscode + Cortex Debug plugin -> openocd -> stlink to write and debug code. Using breakpoints and debug prints via ITM.
+ Wireshark + usbmon to monitor USB flow.
+ [linux-serial-test](https://github.com/cbrake/linux-serial-test) also very useful tool

adadurov, 2020-01-29:
@dmitrystu , thanks for sharing the details of your debugging environment.
Yes, the non-interrupt build for BluePill worked out of the box on my BP device.

xcvista, 2020-01-29:
@adadurov I use macOS and Window 10, so my stack is:

* Eclipse CDT + GNU MCU Eclipse + Segger J-Link for coding and debugging. (You can get a cheap Segger J-Link by reprogramming the ST-Link/V2 on a STM32 Nucleo board.)
* Wireshark for USB sniffing. Additionally you need usbpcap on Windows. (macOS USB sniffing came directly in the kernel.)

commit f9115211 (2017-11-30): test: add interrupt based demo

closes #8