dmitrystu/libusb_stm32 #9: Porting this library to other devices, round two

xcvista, 2017-12-05:
This recommendation is a continuation of the issue #2 but regarding the two USB peripherals found on STM32F2, F4 and F7 families. The existing v2 driver may work for the OTG_FS peripheral, but is there any consideration about the OTG_HS, either in built-in FS PHY mode, or in external HS PHY mode?

I am considering building a STM32F405RG USB 2.0 Hi-Speed Experiment Board for this, with all three USB connections being made available (OTG_FS, OTG_HS with external PHY and OTG_HS with internal FS PHY). Are you interested?

dmitrystu, 2017-12-06:
Yes, it's interesting.
The core can utilize as many USB devices as available. It don't use any static variables.
As far as i know the ST's OTG_HS machine is similar to OTG_FS. But i haven't read the datasheet yet.

xcvista, 2018-01-02:
I have two boards with STM32F407 and '417 up respectively in OTG_FS mode. I will test port this ASAP. Also to be tested is STM32L433, likely will work with some existing driver.

Another few chips I have down the pipeline is F405 (OTG_HS, both internal FS PHY mode and external HS PHY mode, they have the same OTG_FS as F407/F417) F107 (one OTG,) F722 and F756 (both chips, both OTG_FS and OTG_HS.)

kisielk, 2018-01-02:
I have an F413 I'm planning to try this library on, which has just OTG_FS.

xcvista, 2018-01-02:
@kisielk I am on it. I do hope that the F417's OTG_FS is similar enough with F413's so you can just take and use it.

kisielk, 2018-01-02:
@xcvista is it different from the existing v2 driver? or just a matter of adding the necessary defines?

xcvista, 2018-01-02:
@kisielk It is fairly different as how the VBUS detect is handled changed. A lot of stuff in `enable()` and `connect()` are throwing out errors, and I have to remove dependency from a few `#define`'s in `stm32.h` since that header file plainly won't work in my build environment.

Maybe we need to come up with a better naming scheme for things, as the `usb_32vx` will become unmanageable soon after more parts are introduced, especially when `OTG_HS` are being introduced. I already have a few in the pipeline here.

xcvista, 2018-01-17:
I got a Nucleo-F767ZI second hand. So now I have three chips ready to test: STM32F407ZGT6 OTG_FS, STM32F417ZGT6 OTG_FS and STM32F767ZIT6 OTG_FS.

dmitrystu, 2018-01-18:
Fine. I've bought NUCLEO-F429 just before New Year and made otgfs driver. But i have no time for test and debug. I hope i'll do it at weekend.

kisielk, 2018-01-18:
I have an STM32F746G-DISCO I can test on, but don't have time to work on the code at the moment. If someone gives me a basic firmware I can try and help debug though.

xcvista, 2018-01-19:
@dmitrystu
* I will test the stm32f429-otgfs code on STM32F417, F407 and F767 as soon as you put it online. (I spent way too much time on sorting out the platform code and code generators...)
* I have a Chinese workalike of STM32F103, GD32F103CBT6, to test the stm32f103_devfs code on.
* I have a chip report: my two projects based on STM32F042 have been working stable for a while, using the stm32l052_devfs driver. I think you can list STM32F042K6T6 and STM32F042F6P6 as tested. (Yay we covered the cheapest and smallest USB microcontroller available.)

@kisielk I think the same F429 code would work on most F4/F7 chips, hopefully without too much modifications.

dmitrystu, 2018-02-03:
The stm32f429_otgfs driver is ready for testing

FabianInostroza, 2018-09-19:
@dmitrystu I added support for the STM32F105/F107 based on the STM32F429 driver, would you accept a PR?

I also ported the bootloader.

dmitrystu, 2018-09-19:
Sure

PeeJay, 2018-09-24:
Has anyone made progress porting this to an F7 series chip yet? Using the F429 driver and the cdc demo I've got as far as receiving a bus reset event, but nothing after that.

dmitrystu, 2018-09-24:
Check RCC for correct 48MHz USB FS PHY clock setup.

xcvista, 2019-05-26:
Well finally I have an OTG-HS-ULPI board ready.

dmitrystu, 2019-05-26:
Cool !
I've tried to use OTG-HS with Waveshare USB3300+NUCLEO-F429ZI but failed. Perhaps i've had a problems with interconnection wires @60MHz ULPI clock.

xcvista, 2019-05-27:
Let me put the schematics and Gerber on here, and put up a Tindie link if you want to buy one. It is STM32F405RGT6 + USB3300 on one board, with the intention to be used as a CMSIS-DAP debugger with a USB 2.0 Hi-Speed link.

xcvista, 2019-07-30:
Here is my STM32F405 + USB3300 board: https://www.tindie.com/products/maxtch/stm32f405-usb-2-hi-speed-evaluation-board-dap405/