dmitrystu/libusb_stm32 #12: Rename drivers

xcvista, 2018-01-02:
Currently all drivers are named numerically. This will likely get out of hand very soon, especially after `OTG_HS` get introduced. I suggest a rename to all driver names based on the device series and peripheral name, kind of inspired by how Linux kernel name drivers on device trees.

The naming convention would be <tt>usb_*device*_*peripheral*</tt>. The *device* part would be one of the typical chip the driver supports, and the *peripheral* part takes one of the four values:

* `fsdev`: USB Full-Speed Device, as found in STM32F042,
* `otgfs`: USB OTG Full-Speed in Device mode, as found in STM32F407,
* `otghs`: USB OTG High-Speed in Device mode with internal Full-Speed PHY, as found in STM32F407,
* `otghs-ulpi`: USB OTG High-Speed in Device mode with external ULPI High-Speed PHY, as found in STM32F407.

Here is a round-up of renamed current drivers:

* `usb_32v0` -> `usb_stm32f042_fsdev`,
* `usb_32v1` -> `usb_stm32l151_fsdev`,
* `usb_32v2` -> `usb_stm32l476_otgfs`,
* `usb_32v3` -> `usb_stm32f103_fsdev`.

And new names for a few new drivers:

* `usb_stm32f405_otgfs` for OTG_FS on STM32F405/415/407/417
* `usb_stm32f405_otghs` for OTG_HS in internal FS PHY mode on STM32F405/415/407/417
* `usb_stm32f405_otghs_ulpi` for OTG_HS in external HS PHY mode on STM32F405/415/407/417

dmitrystu, 2018-01-02:
The naming template looks good. ```usbd_basedevice_peripheral_options```

xcvista, 2018-01-03:
@dmitrystu Slightly different: <tt>usbd_*basedevice*_*peripheral*_*options*</tt> for driver file names, <tt>USBD_*BASEDEVICE*</tt> as part detect macros, <tt>usbd_*peripheral*_*options*</tt> as names for the structs.

commit 20d81a2b (2018-01-08): Merge branch 'rename'

closes #12
closed 2018-01-08 by dmitrystu