dmitrystu/libusb_stm32 #20: OTG FS TX Fifo allocation

microwavesafe, 2018-04-21:
As you know the OTG FS parts have 1.25KB RAM for the FIFOs. The datasheet says that larger TX FIFOs improve performance, but I noticed your code only assigns the minimum required (epsize or 16 words in set_tx_fifo).
Before I have just split the remaining RAM (after RX and endpoint 0 TX) equally to each of the 3 IN endpoints. There's quite a lot of RAM for only 3 endpoints.
Do you think it is worth having a way for the application code to manually set these TX FIFO sizes?

dmitrystu, 2018-04-21:
For double-buffered bulk and isochronous endpoint double amount of the RAM in TX FIFO will be reserved. Adding the manual TX FIFO configuration will break as USB FS compatibility as other features like dynamic reconfiguration of endpoints. Or will seriously enlarge code size. In some cases it's also critical.
I've reached about 10Mbps (610 kBps for TX channel  and same for RX channel) user-data transfer rate in cdc-loop. It's very close to maximum theoretical speed of the 12Mbps (user-data plus overhead).

microwavesafe, 2018-04-21:
These sound like good reasons not to implement.

closed 2018-04-21 by microwavesafe