dmitrystu/libusb_stm32 #24: HID example request

Serasidis, 2018-06-09:
Congratulations for your great work ! 
I have tested successfully your USB device stack with STM32F103C8 and STM32F407VET6. It works great !

Is there any HID example like the **cdc_loop.c** you have included in the demo folder ?
Any  mouse or keyboard HID example is enough to be used as template for other HID devices

Thanks in advance

Serasidis, 2018-10-12:
4 months with no answer...

closed 2018-10-12 by Serasidis

xcvista, 2018-12-16:
HID is actually a bit complicated, since the keyboard/mouse emulation and generic HID have some differences. Also STM32L4 even supports PS/2 mode for keyboard/mouse HID.

If you want examples of simple HID, I might be able to provide one which is an interrupt-driven USB CDC + HID composite, with the HID part implementing CMSIS-DAP.

lulle2007200, 2020-08-24:
2 years late, but i'd appreciate a composite example.