dmitrystu/libusb_stm32 #27: STM32L433

kosthala, 2018-10-08:
Can i use the library for the STM32L433 controller? 
If yes, what modifications should be made to the library?

dmitrystu, 2018-10-08:
Looks like the USB-FS machine for L433 is the same as for L0, but has a different register names and addresses. Try [L433](https://github.com/dmitrystu/libusb_stm32/tree/L433) experimental branch.

xcvista, 2018-11-24:
I have incorporated it in PR #28.

xcvista, 2018-11-26:
However I am having some problems with STM32L433 here: it won't enumerate. The host machine is not even detecting the USB device, although I have set the GPIO and clocks up correctly.

Meanwhile, the USB bootloader works perfectly.

dmitrystu, 2018-11-26:
I have no ideas why. I have no L433 in my homelab for testing.
What's bootoloader ? Embedded ?

xcvista, 2018-11-27:
@dmitrystu The ROM bootloader that came with the chip can enumerate and operate properly, but this driver won't.

If you need one, I am willing to sell you a STM32L433 kit: https://www.tindie.com/products/maxtch/sushibits-arm-development-kits/

dmitrystu, 2018-11-27:
STM32L433CCT6 looks pin-to-pin compatible with STM32F103C8T6 used on bluepill board. I think i can replace F103 by L433.

dmitrystu, 2018-12-05:
Sometimes ST makes me crazy. Try to set USV bit in PWR->CR2 for L433.

dmitrystu, 2018-12-05:
Here is a my startup code for STM32L433CCT6 on bluepill board. USB FS enumerated OK and cdc-loop works fine. Reached speed about 604 kcps for both TX and RX
````c
    /* using HSI16 as AHB/CPU clock, HSI48 as USB PHY clock */
    _BST(RCC->CR, RCC_CR_HSION);
    _WBS(RCC->CR, RCC_CR_HSIRDY);
    _BMD(RCC->CFGR, RCC_CFGR_SW, RCC_CFGR_SW_HSI);
    _WVL(RCC->CFGR, RCC_CFGR_SWS, RCC_CFGR_SWS_HSI);
    _BST(RCC->CRRCR, RCC_CRRCR_HSI48ON);
    _WBS(RCC->CRRCR, RCC_CRRCR_HSI48RDY);
    _BMD(RCC->CCIPR, RCC_CCIPR_CLK48SEL, 0);
    /* setup PA11 PA12 to AF10 (USB FS) */
    _BST(RCC->AHB2ENR, RCC_AHB2ENR_GPIOAEN);
    _BST(GPIOA->AFR[1], (0x0A << 12) | (0x0A << 16));
    _BMD(GPIOA->MODER, (0x03 << 22) | (0x03 << 24), (0x02 << 22) | (0x02 << 24));
    /* Disabling USB Vddusb power isolation. Vusb connected to Vdd */
    _BST(RCC->APB1ENR1, RCC_APB1ENR1_PWREN);
    _BST(PWR->CR2, PWR_CR2_USV);
````

xcvista, 2018-12-06:
About that speed test, maybe run it on a crystal or turn on CRS with USB SOF or LSE and try again?

My board has no HSE (STM32L432KBU6) so I has to use MSI in LSE PLL mode for that. L433 usually has HSE crystal oscillator so you should be able to run the main PLL on that.

dmitrystu, 2018-12-06:
Speed test is ok. 600K chars per secound for one side is equal 1200K chars per secound on bus (Tx +Rx). And this is about 10M bits per secound.