dmitrystu/libusb_stm32 #31: Moving driver and core entry points into special subsections.

dmitrystu, 2018-12-10:
The main idea to put driver entry points AKA ````const struct usbd_driver```` and core entry point ````void usbd_poll(usbd_device *dev);```` into special named sections like ````.text.usbd_driver```` and ````.text.usbd_core```` respectively.
It will have no effect on general linkage, but can help to put these sections to fixed places in bootloader (just after isr_vectors for instance).
So, because  the  size of ````const struct usbd_driver```` is fixed, the ````usbd_poll(usbd_device *dev)```` entry point also will be on fixed place even bootloader recompiled.
What do you think about this, guys ?

xcvista, 2018-12-16:
No need really. There is a GCC compiler flag `-ffunction-sections`, which achieves the same effect on all functions. That flag is often specified by default along with `--gc-sections` in linker to minimize the binary size by throwing out unused code.

dmitrystu, 2018-12-18:
Yes, it can be done this way. But not for assembly drivers.

commit cb169337 (2018-12-18): add named section for assembly drivers
closes #31
closed 2018-12-18 by dmitrystu