dmitrystu/libusb_stm32 #34: Timeout needed in _WBS

gregaiello, 2019-02-28:
Congratulations on the libs they are amazing and nicely written. I am working with STM32F407 and sometimes the USB_FS peripheral gets stuck (~20% of the times) in the low level driver for the specific microcontroller, specifically when calling the enable() and calling `_WBC(OTG->GRSTCTL, USB_OTG_GRSTCTL_CSRST)`, reading the datasheet the soft core reset is optional (page 1278 of the user reference manual):

> Typically, the software reset is used during software development and also when you
dynamically change the PHY selection bits in the above listed USB configuration registers.
When you change the PHY, the corresponding clock for the PHY is selected and used in the
PHY domain. Once a new clock is selected, the PHY domain has to be reset for proper
operation.

I got rid of that call and I just wait for the idle state `_WBS(OTG->GRSTCTL, USB_OTG_GRSTCTL_AHBIDL);` and everything works fine.

Does anybody have the same issue with the soft core reset or knows if such call is needed?

Thank you again for the efforts.

7134956, 2019-02-28:
I also encountered this problem in some cases, but I have not yet looked for the right solution.
[image]

dmitrystu, 2019-02-28:
Could you please test. For my NUCLEO-F429ZI i see no stucks for both old and new drivers.

7134956, 2019-03-01:
Спасибо большое. Теперь запускается без проблем.

gregaiello, 2019-03-01:
Yes, with the latest commit it works and it is stable on the STM32F407. Thank you!

closed 2019-03-01 by gregaiello