dmitrystu/libusb_stm32 #36: C++ compatibility with usb_string_descriptor & macros

jacobschloss, 2019-05-04:
Hi,

I'm trying to use this in a mixed C & C++ language project for the STM32H750 (looking at porting the otghs driver), and have some issues with usb_string_descriptor and USB_STRING_DESC.

Doing something like
```
static const struct usb_string_descriptor manuf_desc_en = USB_STRING_DESC("Open source USB stack for STM32");`
```
Gives this error, on member wString
```
error: invalid conversion from 'const char16_t*' to 'uint16_t {aka short unsigned int}' [-fpermissive]
                                     .wString = {CAT(u,s)}}
```
If I make wString a pointer, I get a different error, where C++ is pickier about extra braces around scalars

```
error: braces around scalar initializer for type 'uint16_t* {aka short unsigned int*}'
                                     .wString = {__VA_ARGS__}}
```

Changing the macros to this works, also need an explicit cast to uint16_t* so we lose the const-ness

```
/** Macro to create \ref usb_string_descriptor from array */
#define USB_ARRAY_DESC(...)        {.bLength = 2 + sizeof((uint16_t[]){__VA_ARGS__}),\
                                    .bDescriptorType = USB_DTYPE_STRING,\
                                    .wString = (uint16_t*)__VA_ARGS__}
/** Macro to create \ref usb_string_descriptor from string */
#define USB_STRING_DESC(s)         {.bLength = sizeof(CAT(u,s)),\
                                    .bDescriptorType = USB_DTYPE_STRING,\
                                    .wString = (uint16_t*)CAT(u,s)}

struct usb_string_descriptor {
    uint8_t  bLength;
    uint8_t  bDescriptorType;
    uint16_t* wString;
} __attribute__((packed));
```

Would it be ok to change usb_string_descriptor, USB_ARRAY_DESC, and USB_STRING_DESC to something like this? I'm not sure if the brace around the single element matters in this case. I don't think this modifies behavior.

dmitrystu, 2019-05-04:
I think not. According to chapter 9.6.7 of the USB2.0 specification wString must be an array not a pointer.
I will try to find right solution, but i'm not so familiar with C++.

jacobschloss, 2019-05-05:
Ok yeah, I think that would break how sizeof() is used some places to get the descriptor length. I'll just put all the structs in a .c file for now, that seems to compile ok.

closed 2019-06-19 by dmitrystu