dmitrystu/libusb_stm32 #44: stm32f070f6px

bindassdost, 2019-07-21:
i am trying to get your library working on my stm32f070f6 chip from past two days.
i have read previous issue which says stm32l052 code should work on my chipset. beacuse startup code for my chip was missing in cdc_startup.c file. i had added it as below

`#elif defined(STM32F070x6)
    /* set flash latency 1WS */
    _BMD(FLASH->ACR, FLASH_ACR_LATENCY, 1);
    if ((RCC->CFGR & RCC_CFGR_SWS) == RCC_CFGR_SWS_PLL) /* (1) */
    {
        RCC->CFGR &= (uint32_t) (~RCC_CFGR_SW); /* (2) */
        while ((RCC->CFGR & RCC_CFGR_SWS) != RCC_CFGR_SWS_HSI) /* (3) */
        {
            /* For robust implementation, add here time-out management */
        }
    }
    RCC->CR &= (uint32_t)(~RCC_CR_PLLON);/* (4) */
    while((RCC->CR & RCC_CR_PLLRDY) != 0) /* (5) */
    {
        /* For robust implementation, add here time-out management */
    }
    RCC->CFGR = (RCC->CFGR & (~RCC_CFGR_PLLMUL)) | (RCC_CFGR_PLLMUL6); /* (6) */
    RCC->CR |= RCC_CR_PLLON; /* (7) */
    while((RCC->CR & RCC_CR_PLLRDY) == 0) /* (8) */
    {
        /* For robust implementation, add here time-out management */
    }
    RCC->CFGR |= (uint32_t) (RCC_CFGR_SW_PLL); /* (9) */
    while ((RCC->CFGR & RCC_CFGR_SWS) != RCC_CFGR_SWS_PLL) /* (10) */
    {
        /* For robust implementation, add here time-out management */
    }
    RCC->CFGR3 |= RCC_CFGR3_USBSW_PLLCLK;`

but i am still unable to get it working. this clock setup was working with cubemx project so i think my startup code does set and enable usb clock through pll to 48Mhz.

defines that i was using are
        STM32F0, STM32F070x6, USBD_SOF_DISABLED

default linker script for stm32f070f6px was used with --specs=nosys.specs

please help me

mne1991, 2019-07-22:
Please reformat the code, it is unreadable. Do you have an oscilloscope available?

bindassdost, 2019-07-22:
I had fixed formatting the code. I do not have an oscilloscope available. Startup code was taken from stm32f070f6p6 reference manual. same code was tested on CubeMX project where it was working fine but with huge 22KB compile size with only bare-bone CDC loopback. that is why this library is important for me because i had less than 10KB flash available with cubemx

dmitrystu, 2019-07-22:
RCC setup looks correct. According to datasheet PA11 and PA12 must be switched to USB mode automatically. Also this https://github.com/dmitrystu/libusb_stm32/issues/29 maybe helpful

dmitrystu, 2019-07-22:
Checked RM0360. Found 'Figure 11. Clock tree (STM32F070x6, STM32F070xB and STM32F030xC)' doesn't meet '7.4.2 Clock configuration register (RCC_CFGR) bit 16' (HSI vs HSI/2 as PLL in). Try to set PLLMUL to 12.

mne1991, 2019-07-23:
That is exactly why I asked him for an oscilloscope, I had always doubts about configuring the clock to the correct frequency, till I got an oscilloscope and validated the frequency using the MCO pins.

bindassdost, 2019-07-25:
First of all Thank you @dmitrystu for taking time to reply.

i had read reference manual one more time to have bit deeper understanding. i had modified startup code as below but still my usb doesnt get recognized.

```
//
    _BMD(RCC->CR, RCC_CR_HSITRIM, RCC_CR_HSITRIM_4); // SET TRIM TO 16
    _BST(RCC->CR, RCC_CR_HSION); // TURN ON HSI
    _WBS(RCC->CR, RCC_CR_HSIRDY); // WAIT FOR READY
//
    _BMD(RCC->CFGR, RCC_CFGR_SW, RCC_CFGR_SW_HSI); // SWITCH TO HSI
    _WVL(RCC->CFGR, RCC_CFGR_SWS, RCC_CFGR_SWS_HSI); // WAIT FOR HSI SWITCH
//
    _BMD(RCC->CFGR, RCC_CFGR_HPRE, RCC_CFGR_HPRE_DIV1);
    _BMD(RCC->CFGR, RCC_CFGR_PPRE, RCC_CFGR_PPRE_DIV1);
//
    _BST(FLASH->ACR, FLASH_ACR_PRFTBE);// ENABLE PREFETCH
    _BST(FLASH->ACR, FLASH_ACR_LATENCY); // SET LATENCY TO 1 BECAUSE WE WILL SET FREQUENCY TO 48MHz
//
    _BCL(RCC->CR, RCC_CR_PLLON); // TURN OFF PLL
    _WBC(RCC->CR, RCC_CR_PLLRDY); // WAIT FOR PLL OFF
//
    _BMD(RCC->CFGR, RCC_CFGR_PLLSRC, RCC_CFGR_PLLSRC_HSI_DIV2); // SET SOURCE TO HSI/2 = 4MHz
    _BMD(RCC->CFGR, RCC_CFGR_PLLMUL, RCC_CFGR_PLLMUL12); // SET MULTIPLY TO 12 4*12 = 48MHz
//
    _BST(RCC->CR, RCC_CR_PLLON); // TURN ON PLL
    _WBS(RCC->CR, RCC_CR_PLLRDY); // WAIT FOR PLL ON
//
    _BMD(RCC->CFGR, RCC_CFGR_SW, RCC_CFGR_SW_PLL); // SWITCH TO PLL
    _WVL(RCC->CFGR, RCC_CFGR_SWS, RCC_CFGR_SWS_PLL); // WAIT FOR PLL SWITCH
//
    _BMD(RCC->CFGR3, RCC_CFGR3_USBSW, RCC_CFGR3_USBSW_PLLCLK); // SET USB CLOCK TO PLL
    _BST(SYSCFG->CFGR1, SYSCFG_CFGR1_PA11_PA12_RMP); // ENABLE REMAP
    _BST(RCC->APB2ENR, RCC_APB2ENR_SYSCFGCOMPEN);
    _BST(RCC->AHBENR, RCC_AHBENR_GPIOAEN);
    _BST(RCC->APB1ENR, RCC_APB1ENR_USBEN);
```

any suggestion, i am out of clue now

dmitrystu, 2019-07-26:
I have no F070 in my lab, Will order STM32F070RBT6 as a replacement of F103 on bluepill.

microwavesafe, 2019-07-29:
I have got the F070 working with USB, but I don't use the start up code in this project, so it's not an obvious translation.

I did notice you are using the internal oscillator, if the deviation of the internal clock is too high USB won't function as it relies on tight timing tolerances. Being an RC oscillator is depends on voltage and temperature, so the default value your setting RCC_CR_HSITRIM to, may not be correct?

bindassdost, 2019-07-29:
@microwavesafe  its not an oscillator issue because cubemx code with internal rc oscillator works fine but is too heavy

microwavesafe, 2019-07-29:
As mentioned previously the only additional setup I added was ```RCC->CFGR3 |= RCC_CFGR3_USBSW;``` which you have already covered.

Are you getting nothing at all on the PC end? It's it attempting to enumerate the device? Do you get any USB irq events in the F070? Can you use the debugger to prove this? Giving us some more details on what is not working may help us narrow down the problem.

bindassdost, 2019-07-29:
Yes device tries to enumerate and I had USB device not recognized error=43, I had only STLink-v2 debugger and I don't know how to debug with that.

microwavesafe, 2019-07-29:
If it's trying to enumerate then it is likely not a hardware register issue, if it was you tend to get nothing at all at the PC.

Did you match the VID, PID values to the same as the CubeMX demo?

microwavesafe, 2019-07-29:
By the way you can use VSCode + cortex debug, for debugging.

https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug

I believe it supports stlink, via Texane's st-util GDB server

bindassdost, 2019-07-29:
i will try that and report here but error 43 mean failed to read device descriptor so I thought it was a hardware issue. anyway thanks for guidance.

microwavesafe, 2019-07-29:
That's only in my experience. I don't use windows, so I'm not familiar with the error numbers. Can you use a windows based tool to determine at which point the enumeration fails? That could help as well.

microwavesafe, 2019-07-29:
Here are my RCC registers for a project based on an F070RB with working USB. Bare in mind I am using an external oscillator at 12MHz, so they will be different. However it may still be useful

CR: 0x3035983
CFGR: 0x9000a
CIR: 0x0
APB2RSTR: 0x0
APB1RSTR: 0x0
AHBENR: 0x5e0014
APB2ENR: 0x5001
APB1ENR: 0xc00000
BDCR: 0x811b
CSR: 0x1c800000
AHBRSTR: 0x0
CFGR2: 0x0
CFGR3: 0x80
CR2: 0x7580

dmitrystu, 2019-07-31:
Tested with F070CBT6 on the bluepill board w/o R10 (DP pull-up)

dmitrystu, 2019-07-31:
Also tested with L052 ASM driver for F070. It works fine.

closed 2019-08-05 by dmitrystu