dmitrystu/libusb_stm32 #45: CDC and Serial devices does work on Windows (STM32F429)

mne1991, 2019-07-23:
Hello,

I flashed the code STM32F429 MCU, the HID (mouse) works fine (tested on Linux and Windows). However, CDC and Serial port devices don't work (tested on WIndows) and they are shown with a yellow exclamation mark.

Here are the information I got from windows device manager.

### CDC Device:

Status:

> The drivers for this device are not installed. (Code 28)
There are no compatible drivers for this device.

Events:
 
> Device USB\VID_0483&PID_5740&MI_01\7&118ad3aa&0&0001 was configured.
Driver Name: null
Class Guid: {00000000-0000-0000-0000-000000000000}
Driver Date: 
Driver Version: 
Driver Provider: 
Driver Section: 
Driver Rank: 0x0
Matching Device Id: 
Outranked Drivers: 
Device Updated: false
Parent Device: USB\VID_0483&PID_5740\5682CFA2

### Serial port:

Status:

> This device cannot start. (Code 10)
A device which does not exist was specified.

Events:

> Driver Management concluded the process to install driver usbser.inf_amd64_6821f829c6b22a54 for Device Instance ID USB\VID_0483&PID_5740&MI_00\7&118AD3AA&0&0000 with the following status: 0x0.

> Driver Management has concluded the process to add Service usbser for Device Instance ID USB\VID_0483&PID_5740&MI_00\7&118AD3AA&0&0000 with the following status: 0.

> Device USB\VID_0483&PID_5740&MI_00\7&118ad3aa&0&0000 requires further installation.

dmitrystu, 2019-07-27:
Using usblyzer found that device doesn't answer to the URB_GET_STATUS_FROM_DEVICE request.

mne1991, 2019-07-27:
Would you fix it, or do you suggest any concrete fix?

dmitrystu, 2019-07-28:
Have you tried with last commit? It resolves YB for CDC under WIN10.
See ref https://github.com/dmitrystu/libusb_stm32/pull/30

mne1991, 2019-07-28:
Thank you for the commit. I haven't tried it yet. At the moment I am trying to minimize the code as far as possible to be able to understand it (it is actually relatively complex for me as a beginner in USB driver world).

I will try it tomorrow and give you a feedback.

mne1991, 2019-08-01:
Sorry for being late. Are you sure that it is working? No devices are being detected now, as if I didn't plug any device to USB.

dmitrystu, 2019-08-01:
Here is an usbview report
````
[Port2]  :  Составное USB устройство

Is Port User Connectable:         yes
Is Port Debug Capable:            no
Companion Port Number:            14
Companion Hub Symbolic Link Name: USB#ROOT_HUB30#4&369d7e28&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
Protocols Supported:
 USB 1.1:                         yes
 USB 2.0:                         yes
 USB 3.0:                         no

Device Power State:               PowerDeviceD0

       ---===>Device Information<===---
English product name: "CDC Loopback demo"

ConnectionStatus:                  
Current Config Value:              0x01  -> Device Bus Speed: Full (is not SuperSpeed or higher capable)
Device Address:                    0x05
Open Pipes:                           4

          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0200
bDeviceClass:                      0x00
*!*ERROR: device class should be Multi-interface Function 0xEF
          When IAD descriptor is used
bDeviceSubClass:                   0x00
*!*ERROR: device SubClass should be USB Common Sub Class 2
          When IAD descriptor is used
bDeviceProtocol:                   0x00
*!*ERROR: device Protocol should be USB IAD Protocol 1
          When IAD descriptor is used
bMaxPacketSize0:                   0x08 = (8) Bytes
idVendor:                        0x0483 = STMicroelectronics
idProduct:                       0x5740
bcdDevice:                       0x0100
iManufacturer:                     0x01
     English (United States)  "Open source USB stack for STM32"
iProduct:                          0x02
     English (United States)  "CDC Loopback demo"
iSerialNumber:                     0xFE
     English (United States)  "E15E4D9F"
bNumConfigurations:                0x01

          ---===>Open Pipes<===---

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x82  -> Direction: IN - EndpointID: 2
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0008 = 0x08 bytes
bInterval:                         0xFF

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x83  -> Direction: IN - EndpointID: 3
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0010 = 0x10 bytes
bInterval:                         0x32

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x01  -> Direction: OUT - EndpointID: 1
[... 108 lines cut ...]
bAlternateSetting:                 0x00
bNumEndpoints:                     0x01
bInterfaceClass:                   0x03  -> HID Interface Class
bInterfaceSubClass:                0x00
bInterfaceProtocol:                0x00
iInterface:                        0x00

          ===>HID Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x21
bcdHID:                          0x0100
bCountryCode:                      0x00
bNumDescriptors:                   0x01
bDescriptorType:                   0x22 (Report Descriptor)
wDescriptorLength:               0x0030

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x83  -> Direction: IN - EndpointID: 3
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0010 = 0x10 bytes
bInterval:                         0x32
````
OOPS. Some errors detected but no YB for the device in the device manager

mne1991, 2019-08-01:
How did you detect those errors? I am very interested to know how you debug your USB firmware (as I cannot afford a $1000 hardware analyzer).

dmitrystu, 2019-08-01:
usbview is a part of the windows-10-sdk (debug)
usblyzer also is very useful software for Windows. (30 days free trial)
But i prefer the usbmon+wireshark under Linux.

mne1991, 2019-08-01:
Thank you! I use the same software also.

Do you have any idea how to solve that?

Actually this part of bugs is very advance (at least for me), as I said before, I am very new to USB and trying to read your library and try to understand every details in it. I am going forward very very slowly (10 lines per day), but at least I am moving.

dmitrystu, 2019-08-01:
Try to set BKP at the cdc_getdesc entry point and inspect request data.

mne1991, 2019-08-01:
The descriptor fix did not help, I will try tomorrow when I come back home to set a break point and inspect the request data.

Actually setting break point during debugging USB is a pain, because the protocol is very time strict, so every time I need to reset the program and start over and over. Do you have any tip to debug it more efficiently?

dmitrystu, 2019-08-01:
Typical USB enumeration process is
```
Reset -> Probe Device Descriptor -> Set Address -> Read Device Descriptor -> Read Device Qualifier -> Read Configuration Descriptor -> Read Strings -> Set Configuration
```

mne1991, 2019-08-16:
Excuse me till now I were not able to solve the issue, I am still analyzing the whole code trying to understand it. Could you take another look on it? maybe you find something which can solve the issue.

dmitrystu, 2019-08-17:
Checked from latest master on my NUCLEO-F429ZI. Found no problems with both FS and HS(in FS mode) under Linux and Win10.
````
No.     Time           Source                Destination           Protocol Length Info
     47 6.250988       host                  2.2.0                 USBHUB   64     GET_STATUS Request     [Port 5]
     48 6.251078       2.2.0                 host                  USBHUB   68     GET_STATUS Response    [Port 5]
     49 6.251089       host                  2.2.0                 USBHUB   64     CLEAR_FEATURE Request  [Port 5: C_PORT_CONNECTION]
     50 6.251198       2.2.0                 host                  USBHUB   64     CLEAR_FEATURE Response [Port 5: C_PORT_CONNECTION]
     51 6.251208       host                  2.2.0                 USBHUB   64     GET_STATUS Request     [Port 5]
     52 6.251324       2.2.0                 host                  USBHUB   68     GET_STATUS Response    [Port 5]
     53 6.286131       host                  2.2.0                 USBHUB   64     GET_STATUS Request     [Port 5]
     54 6.286205       2.2.0                 host                  USBHUB   68     GET_STATUS Response    [Port 5]
     55 6.322125       host                  2.2.0                 USBHUB   64     GET_STATUS Request     [Port 5]
     56 6.322204       2.2.0                 host                  USBHUB   68     GET_STATUS Response    [Port 5]
     57 6.358121       host                  2.2.0                 USBHUB   64     GET_STATUS Request     [Port 5]
     58 6.358191       2.2.0                 host                  USBHUB   68     GET_STATUS Response    [Port 5]
     59 6.394129       host                  2.2.0                 USBHUB   64     GET_STATUS Request     [Port 5]
     60 6.394222       2.2.0                 host                  USBHUB   68     GET_STATUS Response    [Port 5]
     61 6.394248       host                  2.2.0                 USBHUB   64     SET_FEATURE Request    [Port 5: PORT_RESET]
     62 6.394349       2.2.0                 host                  USBHUB   64     SET_FEATURE Response   [Port 5: PORT_RESET]
     64 6.414193       2.2.0                 host                  USBHUB   68     GET_STATUS Response    [Port 5]
     65 6.414206       host                  2.2.0                 USBHUB   64     CLEAR_FEATURE Request  [Port 5: C_PORT_RESET]
     66 6.414314       2.2.0                 host                  USBHUB   64     CLEAR_FEATURE Response [Port 5: C_PORT_RESET]
     67 6.474146       host                  2.0.0                 USB      64     GET DESCRIPTOR Request DEVICE
     68 6.474346       2.0.0                 host                  USB      72     GET DESCRIPTOR Response DEVICE[Malformed Packet]
     69 6.474365       host                  2.2.0                 USBHUB   64     SET_FEATURE Request    [Port 5: PORT_RESET]
     70 6.474473       2.2.0                 host                  USBHUB   64     SET_FEATURE Response   [Port 5: PORT_RESET]
     71 6.494137       host                  2.2.0                 USBHUB   64     GET_STATUS Request     [Port 5]
     72 6.494228       2.2.0                 host                  USBHUB   68     GET_STATUS Response    [Port 5]
     73 6.494254       host                  2.2.0                 USBHUB   64     CLEAR_FEATURE Request  [Port 5: C_PORT_RESET]
     74 6.494348       2.2.0                 host                  USBHUB   64     CLEAR_FEATURE Response [Port 5: C_PORT_RESET]
     75 6.554127       host                  2.0.0                 USB      64     SET ADDRESS Request
     76 6.554227       2.0.0                 host                  USB      64     SET ADDRESS Response
     77 6.582131       host                  2.26.0                USB      64     GET DESCRIPTOR Request DEVICE
     78 6.582353       2.26.0                host                  USB      82     GET DESCRIPTOR Response DEVICE
     79 6.582373       host                  2.26.0                USB      64     GET DESCRIPTOR Request DEVICE QUALIFIER
     80 6.582478       2.26.0                host                  USB      64     GET DESCRIPTOR Response
     81 6.582494       host                  2.26.0                USB      64     GET DESCRIPTOR Request DEVICE QUALIFIER
     82 6.582852       2.26.0                host                  USB      64     GET DESCRIPTOR Response
     83 6.582869       host                  2.26.0                USB      64     GET DESCRIPTOR Request DEVICE QUALIFIER
     84 6.583195       2.26.0                host                  USB      64     GET DESCRIPTOR Response
     85 6.583214       host                  2.26.0                USB      64     GET DESCRIPTOR Request CONFIGURATION
     86 6.583567       2.26.0                host                  USB      73     GET DESCRIPTOR Response CONFIGURATION
     87 6.583575       host                  2.26.0                USB      64     GET DESCRIPTOR Request CONFIGURATION
     88 6.583942       2.26.0                host                  USB      164    GET DESCRIPTOR Response CONFIGURATION
     89 6.583954       host                  2.26.0                USB      64     GET DESCRIPTOR Request STRING
     90 6.584099       2.26.0                host                  USB      68     GET DESCRIPTOR Response STRING
     91 6.584116       host                  2.26.0                USB      64     GET DESCRIPTOR Request STRING
[... cut ...]
    106 6.587442       2.26.0                host                  USBHID   112    GET DESCRIPTOR Response HID Report
    107 6.587870       host                  2.26.3                USB      64     URB_INTERRUPT in
    108 6.609196       2.26.3                host                  USB      64     URB_INTERRUPT in
    109 6.609202       host                  2.26.3                USB      64     URB_INTERRUPT in
    110 6.641196       2.26.3                host                  USB      67     URB_INTERRUPT in
````
Pkts 67 and 68 it's a device probing.

mne1991, 2019-08-17:
So everything works properly with you?

dmitrystu, 2019-08-17:
AFAIK this is the part of the enumeration process. Probing device before address setup.

mne1991, 2019-08-18:
Oh! The devices are there now, it was my fault apparently, I just deleted everything and recreated the project and everything now is recognizable.

Actually I am not able to test the Com loop back device, how may I test it? I opened a *Putty* session to the corresponding virtual Com port, but what to do now? I tried to write something in the console but nothing happened. What should I do exactly?

dmitrystu, 2019-08-18:
This is not a real FIFO. Just only loopback buffer. 
````
RX Host data A -> Dev: A|XXXXXXX
RX Host data B -> Dev: AB|XXXXXX
TX Dev data -> Host:   B|XXXXXXX  Data A transfer
RX Host data C -> Dev: BC|XXXXXX
RX Host data D -> Dev: BCD|XXXXX
TX Dev data -> Host:   CD|XXXXXX  Data B transfer
TX Dev data -> Host:   D|XXXXXXX  Data C transfer
````

mne1991, 2019-08-18:
I will close the issue as the problem has been resolved in your last commit.

Just about my last question, I meant reading and writing to the FIFO generally in `ep_write` and `ep_read` functions respectively:

[This part](https://github.com/dmitrystu/libusb_stm32/blob/26bad84152062c590bd7a694cada47a72e07d34e/src/usbd_stm32f429_otghs.c#L365):

    while (_len--) {
        *_fifo = *(__attribute__((packed)) uint32_t*)buf;
        buf += 4;
    }

and [this part](https://github.com/dmitrystu/libusb_stm32/blob/26bad84152062c590bd7a694cada47a72e07d34e/src/usbd_stm32f429_otghs.c#L332):

    for (unsigned i = 0; i < len; i +=4) {
        uint32_t _t = *fifo;
        if (blen >= 4) {
            *(__attribute__((packed))uint32_t*)buf = _t;
            blen -= 4;
            buf += 4;
        }

As seen from above, the pointer to the FIFO is not being moved at all. How does this work? as I expect, in `read` case, by reading the data are being popped and shifted one address back to the very first address in the FIFO automatically. In `write` case, as long the endpoint is enabled, the data written to the first address will be immediately sent and will not be stacked or kept in the FIFO (i.e. when I write then to the very first address, they will be sent and removed from the FIFO), are those mechanisms correct? If yes or no, could you tell me please where I can find this mechanism in the reference manual or in any other resource?

closed 2019-08-18 by mne1991

dmitrystu, 2019-08-18:
You can read about USB-OTG hardware FIFO's in the corresponded datasheet. For instance RM0090, Article 34.10 USB data FIFOs.

mne1991, 2019-08-19:
I did it already, though the point I mentioned is not well explained in the reference manual. Anyway, it is a different topic and I do not want to go further from the topic of this thread. Thank you!