dmitrystu/libusb_stm32 #46: New chip request: STM32F446 and STM32F732.

xcvista, 2019-09-01:
Those chips are some of the newer high performance chips, and for STM32F446 I am planning to use it in a project. They uses an updated OTG block so existing STM32F405 code will likely not work.

STM32F446 is similar to STM32F405 that it has an `OTG_FS` block with internal PHY, as well as an `OTG_HS` block with internal FS PHY and ULPI pins.

STM32F732 is part of the new STM32F7 line. The `OTG_FS` block is similar to that of STM32F446. Its `OTG_HS` block features internal HS PHY and ULPI pins.

For STM32F446, it need three drivers: `otgfs`, `otghs` and `otghs_ulpi`. For STM32F732 it needs only `otgfs` and `otghs`. Since the internal PHY for `OTG_HS` in STM32F732 is already a HS PHY, there is no point except for compatibility reasons to use external PHY at all, so that more-or-less redundant feature can be left unsupported.

dmitrystu, 2019-09-01:
Maybe I missed something but:
+ STM32F446 don't have an internal HS PHY (see: DocID027107 Rev 6)
+ STM32F732 also don't have an internal HS PHY, but F733 has. (see: DS11854 Rev 4)

xcvista, 2019-09-02:
@dmitrystu My bad.

I wonder how should I modify the files to add support for those chips?

I have created two boards, one with STM32F405 + USB3300 for `stm32f429_otghs_ulpi`, one with STM32F446 + USB3343 + USB mux for `stm32f446_otgfs` and `stm32f446_otghs_ulpi`. And I have a STM32F767 Nucleo that can allow work on `stm32f767_otgfs`.

closed 2020-05-21 by dmitrystu