dmitrystu/libusb_stm32 #48: No seperate callback for in/out endpoint possible

Seneral, 2020-04-04:
The [demo makes clear](https://github.com/dmitrystu/libusb_stm32/blob/master/demo/cdc_loop.c#L439) that the following is expected to work:
```
usbd_reg_endpoint(dev, CDC_RXD_EP, cdc_rxonly);
usbd_reg_endpoint(dev, CDC_TXD_EP, cdc_txonly);
```
(where CDC_RXD_EP = 0x01, CDC_TXD_EP = 0x81).
However, [when the callbacks are stored](https://github.com/dmitrystu/libusb_stm32/blob/master/inc/usbd_core.h#L400), the in/out distinction is thrown away:
```
dev->endpoint[ep & 0x07] = callback;
```
Thus resulting in both endpoints having the same callback, despite expected to have their own.

Not entirely sure about this, how this relates to the endpoint limitation noted in the readme, since I haven't finished porting my code from HAL USB stack. But from the code I see no reason to not store and access the callbacks as follows:
```
dev->endpoint[((ep >> 4) & 0x08) | (ep & 0x07)]
```

For me not a big deal, as a workaround I can different base is for the in/out endpoints

dmitrystu, 2020-04-05:
Yes. There is one callback for each physical endpoint. I think this is a reasonable simplification. Another simplification is equality between physical and logical numbers for the endpoints. In the case of using a combined endpoint, TX/RX flow can be selected in accordance with the event passed into the callback.

dmitrystu, 2020-04-05:
Also, the number of the endpoint passed into callback is a full logical endpoint number with direction bit.

dmitrystu, 2020-04-05:
Yes. There is a bug in the demo code. I have changed endpoints with https://github.com/dmitrystu/libusb_stm32/commit/c280f0649ee2d1c8d83427a28f2f9d27083eac70 because of no enough endpoints in F4 otgfs. It will be fixed.

Seneral, 2020-04-05:
Alright, thanks for the clarification. Finished porting and much happier with this library compared to HAL USB stack.

closed 2020-05-09 by dmitrystu