dmitrystu/libusb_stm32 #49: PlatformIO support

dzarda, 2020-04-07:
Major kudos for the library.

Simply including the repo in the PlatformIO project manifest doesn't work out of the box because of different `inc` vs `include` convention:

```ini
[env]
lib_deps =
    dmitrystu/libusb_stm32
```

We can create a library.json manifest in the repo to enable this. What do you think?

dmitrystu, 2020-04-07:
No problem. Send the PR. It will be merged.

dzarda, 2020-04-13:
Thank you. It will require somehow handling the dependencies. E.g `stm32.h` uses a long path in its includes as opposed to only having "stm32fxx.h" (PlatformIO wants that for `ststm32`). I'll think about a clean way around this.

For now I have simply copy pasted the thing into my project.

dmitrystu, 2020-04-13:
This is a simply wrapper for the generic CMSIS includes provided by STM. I didn't find a solution like ```#include<avr/io.h>``` for the AVR, so i made it.

dmitrystu, 2020-04-13:
There are some utility macros in this include. I can move them to the library code.

dzarda, 2020-04-14:
Why is this full path needed? `"STM32F1xx/Include/stm32f1xx.h"` I've never seen it used like this. From my experience `-I` always points to Include/.

Anyway having it part of libusb OOTB would be slightly easier not only for PIO consumers but everyone.

dmitrystu, 2020-04-14:
Just followed [this guideline](https://github.com/ARM-software/CMSIS_5/tree/develop/Device/_Template_Vendor).

dmitrystu, 2020-04-14:
I use [CMSIS 5](https://github.com/ARM-software/CMSIS_5) with the includes from the ST git repo. Hmm. Looks they's can be got as submodules now.

dzarda, 2020-04-22:
- AFAICS the CMSIS guide makes no statement about `-I` location (sadly).
- PlatformIO (ststm32) points the`-I` to the `stm32f1xx.h` directly

dmitrystu, 2020-04-22:
Why? Looks like there is a pretty good described structure:
```
<cmsis_root>
    + CMSIS
    + Device
       + <Vendor>
          +<Device>
            + Include
            + Source
               + ARM
               + GCC
               + IAR
```
BTW, Which define/attribute/other_mark usually used in the PlatformIO to determine what kind of device-specific code must be used ?

dzarda, 2020-04-26:
The structure is specified well, but they don't tell you where to point the includePath.

PlatformIO expects this to work: `#include "stm32f1xx.h"`

dzarda, 2021-05-17:
For completeness - support would require selecting the appropriate port source based on the STMXXX macros. Back then I got sorta stuck on that, but should be doable.

jcw, 2022-01-28:
Old issue, but allow me to chime in:

As far as I can tell, the defines set in PlatformIO for STM32 boards are a bit of a mystery, what I always end up doing in my projects is to define the `STM32F1` macro etc myself, using `build_flags = -DSTM32F1` in my `platformio.ini` files.

As for `stm32.h`, I think the following alternative version would work with PIO - from this:
```
#elif defined(STM32F1)
    #include "STM32F1xx/Include/stm32f1xx.h"
```
... to:
```
#elif defined(STM32F1)
    #include "stm32f1xx.h"
```
(for all chip families, of course)

When `framework = cmsis` is set in `platformio.ini`, then the include path will automatically be set by PIO, and point to the CMSIS headers which it manages itself in `~/.platformio/packages/framework-cmsis-STM32F1/` etc.

The only changes involved, as far as I can tell, are: 1) to add a `library.json` file (see [docs](https://docs.platformio.org/en/latest/librarymanager/config.html)), 2) to include a modified `stm32.h` for PIO use, and 3) to [publish](https://docs.platformio.org/en/latest/core/userguide/package/cmd_publish.html#cmd-package-publish) this library so PIO will find it. Then, all anyone has to do is to add `lib_deps = libusb_stm32` to their own `platformio.ini` file and this library will get auto-installed from this `git` repository, built, and linked-in.

I'm new to `libusb_stm32`, but I'll see if I can get a demo working for the Blue Pill.

jcw, 2022-01-28:
Ok, I think this will do the trick:

- unpack this zip file in repo root, this creates a `demo-pio/` folder: [demo-pio.zip](https://github.com/dmitrystu/libusb_stm32/files/7961733/demo-pio.zip)
- create a `library.json` file in repo root with this content:

```
{
    "name": "libusb_stm32",
    "description": "Lightweight USB Device Stack",
    "version": "1.0.0",
    "keywords": "stm32, usb",
    "authors": {
        "name": "Dmitry Filimonchuk",
        "email": "dmitrystu@gmail.com"
    },
    "repository": {
        "type": "git",
        "url": "https://github.com/dmitrystu/libusb_stm32.git"
    },
    "frameworks": [ "cmsis", "stm32cube" ],
    "platforms": [ "ststm32" ],
    "headers": "usb.h",
    "build": { "includeDir": "inc" }
}
```

That's basically it, but if you don't have PlatformIO, you also need to install that. The simplest is the command-lin version, see [this page](https://docs.platformio.org/en/latest/core/installation.html). Then from the shell (I'm assuming MacOS or Linux, have no experience with Windows), type the following:

```
cd demo-pio
pio run -t upload
```

The upload step will only succeed if a Blue Pill with ST-Link is attached, evidently.

- I had to alter the demo code so that `SystemInit()` is not redefined, as it's already in the CMSIS code, and sets up some basics (not the clock, just stack and FPU, IIRC) - my solution was to copy the `cdc_*.c` into one `cdc_main.c` and change the main init sequence slightly
- `library.json` has all the info needed to define `libusb_stm32` as a library which PIO understands - it's not published or installed anywhere, but the `demo-pio` project loads the library as if it was (see `platformio.ini`)
- if you want to publish this in the PIO way, you'll need to use that `pio publish` command I mentioned earlier (which requires  setting up an account)
- ... but there's no need: even as it is, people can request this library from GitHub, by adding `lib_deps =  dmitrystu/libusb_stm32` to their own `platformio.ini` project file - this will work once the `library.json` file is present on GitHub

There are probably some details I missed, but I think this library would be very nice to have available in any PIO project.

PR #115 (2022-08-30): Add PlatformIO support
Regarding `stm32_compat.h`: since using both [stm32h](https://github.com/dmitrystu/stm32h) and pio at the same time would lead to code duplication, I thought this would be a good middle ground to have both Makefile and pio supported at the same time.

Closes #49.
closed 2022-08-30 by dmitrystu