dmitrystu/libusb_stm32 #51: CDC OUT ep callback invocation

dzarda, 2020-04-21:
I'm having trouble with the CDC TX. The callback isn't called after not performing any writes in the previous invocation.

The functionality is described in #1 and you suggest writing an empty message to notify the callback. at start (demo has this). But if the callback doesn't initiate another `write` then the next callback isn't invoked. I tried busy-notifying the ep whenever I there's data, but it causes weird data scrambling. I'm calling `poll` in a busy loop.

How is one supposed to know when to perform the notification?

dzarda, 2020-04-21:
Also is it valid to not to use the callback mechanism for TX and `write` from outside the lib? (same context)

dmitrystu, 2020-04-22:
TX callback invoked only once when the EP->HOST transfer completed. But you can call ```usbd_hw_ep_write(ep, data, size)``` at any time to store data in the EP for the next transfer. -1 will be returned in case of unconfigured or busy EP, the number of the stored bytes otherwise.

dmitrystu, 2020-04-22:
Note. There is a bug in the demo code. https://github.com/dmitrystu/libusb_stm32/issues/48 Not yet fixed.

dzarda, 2020-04-26:
Thanks! I rewrote the thing to use endpoint polling. Works, but the data loss/overrun issue is still present. If I send more than 64b in a single Teraterm copy paste, the first 64 bytes are lost. It's perfectly possible I'm screwing up elsewhere tho.

```cpp
void tunnel_downstream_handler() {
    if (primary_uart_tx_ready()) {
        int transferred = usbd_ep_read(&udev, CDC_RXD_EP, PrimaryTxBuf.data(), PrimaryTxBuf.size());
        if (transferred > 0) {
            primary_uart_tx(PrimaryTxBuf.data(), transferred);
        }
    }
}

void tunnel_upstream_handler() {
    primary_uart_rx_poll();
    auto readable = PrimaryRxFifo.readable_range();
    if (readable.second > 0) {
        int transferred = usbd_ep_write(&udev, CDC_TXD_EP, readable.first, std::min(readable.second, size_t(CDC_DATA_SZ)));
        if (transferred > 0) {
            PrimaryRxFifo.notify_read(transferred);
        }
    }
}
```

dmitrystu, 2020-04-28:
See no data loss on the test with 100Meg payload. Tested on STM32F103 bluepill with following script (passing huge gzipped file trough loopback).
```
stty -F $1 0:0:18b2:0:0:0:0:0:0:5:80:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
dd if=test.tar.gz of=$1 bs=1k &
tar -tzf $1
```
What is the configuration of the TXD_EP ?

dzarda, 2020-05-02:
The config is basically the same as demo - single buffer:

```c
.data_eptx = {
        .bLength = sizeof(struct usb_endpoint_descriptor),
        .bDescriptorType = USB_DTYPE_ENDPOINT,
        .bEndpointAddress = CDC_TXD_EP,
        .bmAttributes = USB_EPTYPE_BULK,
        .wMaxPacketSize = CDC_DATA_SZ,
        .bInterval = 0x01,
    },
...

        usbd_ep_config(dev, CDC_TXD_EP, USB_EPTYPE_BULK, CDC_DATA_SZ);
```

I'm calling my logic in a tight busy loop.

dzarda, 2020-05-03:
This is weird, when I copy-paste a paragraph of text, Windows apparently sends a frame with a massive length. Why? I thought endpoint size was limited to 64b here.

<details>
  <summary>Frame from Wireshark</summary>

#### Parsed:

```
Frame 127519: 296 bytes on wire (2368 bits), 296 bytes captured (2368 bits) on interface wireshark_extcap2040, id 0
    Interface id: 0 (wireshark_extcap2040)
    Encapsulation type: USB packets with USBPcap header (152)
    Arrival Time: May  3, 2020 23:41:25.686617000 Central Europe Daylight Time
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1588542085.686617000 seconds
    [Time delta from previous captured frame: 0.003582000 seconds]
    [Time delta from previous displayed frame: 41.030842000 seconds]
    [Time since reference or first frame: 394.074154000 seconds]
    Frame Number: 127519
    Frame Length: 296 bytes (2368 bits)
    Capture Length: 296 bytes (2368 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: usb]
USB URB
    [Source: host]
    [Destination: 6.2.1]
    USBPcap pseudoheader length: 27
    IRP ID: 0xffff8083863492b0
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER (0x0009)
    IRP information: 0x00, Direction: FDO -> PDO
        0000 000. = Reserved: 0x00
        .... ...0 = Direction: FDO -> PDO (0x0)
    URB bus id: 6
    Device address: 2
    Endpoint: 0x01, Direction: OUT
        0... .... = Direction: OUT (0)
        .... 0001 = Endpoint number: 1
    URB transfer type: URB_BULK (0x03)
    Packet Data Length: 269
    [Response in: 127520]
    [bInterfaceClass: CDC-Data (0x0a)]
Leftover Capture Data: 54582063616c6c6261636b20696e766f6b6564206f6e6c79…
```

#### Unparsed:
```
0000   1b 00 b0 92 34 86 83 80 ff ff 00 00 00 00 09 00   ....4...........
0010   00 06 00 02 00 01 03 0d 01 00 00 54 58 20 63 61   ...........TX ca
0020   6c 6c 62 61 63 6b 20 69 6e 76 6f 6b 65 64 20 6f   llback invoked o
0030   6e 6c 79 20 6f 6e 63 65 20 77 68 65 6e 20 74 68   nly once when th
0040   65 20 45 50 2d 3e 48 4f 53 54 20 74 72 61 6e 73   e EP->HOST trans
0050   66 65 72 20 63 6f 6d 70 6c 65 74 65 64 2e 20 42   fer completed. B
0060   75 74 20 79 6f 75 20 63 61 6e 20 63 61 6c 6c 20   ut you can call 
0070   65 70 5f 77 72 69 74 65 28 65 70 2c 20 64 61 74   ep_write(ep, dat
0080   61 70 74 72 2c 20 73 69 7a 65 29 20 61 74 20 61   aptr, size) at a
0090   6e 79 20 74 69 6d 65 20 74 6f 20 73 74 6f 72 65   ny time to store
00a0   20 64 61 74 61 20 69 6e 20 74 68 65 20 45 50 20    data in the EP 
00b0   66 6f 72 20 74 68 65 20 6e 65 78 74 20 74 72 61   for the next tra
00c0   6e 73 66 65 72 2e 20 2d 31 20 77 69 6c 6c 20 62   nsfer. -1 will b
00d0   65 20 72 65 74 75 72 6e 65 64 20 69 6e 20 63 61   e returned in ca
00e0   73 65 20 6f 66 20 75 6e 63 6f 6e 66 69 67 75 72   se of unconfigur
00f0   65 64 20 6f 72 20 62 75 73 79 20 45 50 2c 20 74   ed or busy EP, t
0100   68 65 20 6e 75 6d 62 65 72 20 6f 66 20 74 68 65   he number of the
0110   20 73 74 6f 72 65 64 20 62 79 74 65 73 20 6f 74    stored bytes ot
0120   68 65 72 77 69 73 65 2e                           herwise.
```

</details>

dmitrystu, 2020-05-05:
It's a URB packet, not a capture from a physical interface. It's not split by chunks yet and the length of this packet is a system-depended. AFAIK there is no way to capture chunks without a logic analyzer or sniffer, but I'm not sure about usblyzer. Looks like you received a zero-length packet. I have no idea why. Is the demo code works for you?

This what i have got on my side. EP is 64-byte wide.
````
54	7.100714	host	2.23.1	USB	509	URB_BULK out
55	7.100719	host	2.23.1	USB	66	URB_BULK out
56	7.101132	2.23.1	host	USB	192	URB_BULK in
57	7.101140	host	2.23.1	USB	64	URB_BULK in
58	7.101511	2.23.1	host	USB	192	URB_BULK in
59	7.101516	host	2.23.1	USB	64	URB_BULK in
60	7.101761	2.23.1	host	USB	64	URB_BULK out
61	7.101766	2.23.1	host	USB	192	URB_BULK in
62	7.101770	host	2.23.1	USB	64	URB_BULK in
63	7.101847	2.23.1	host	USB	64	URB_BULK out
64	7.101852	2.23.1	host	USB	125	URB_BULK in
65	7.101854	host	2.23.1	USB	64	URB_BULK in
66	7.101968	host	2.23.0	USB	64	URB_CONTROL out
````
<details>
<summary>OUT Endpoint</summary>

````
Frame 54: 509 bytes on wire (4072 bits), 509 bytes captured (4072 bits) on interface 0
USB URB
    [Source: host]
    [Destination: 2.23.1]
    URB id: 0xffff9b9a828decc0
    URB type: URB_SUBMIT ('S')
    URB transfer type: URB_BULK (0x03)
    Endpoint: 0x01, Direction: OUT
        0... .... = Direction: OUT (0)
        .... 0001 = Endpoint number: 1
    Device: 23
    URB bus id: 2
    Device setup request: not relevant ('-')
    Data: present (0)
    URB sec: 1588689876
    URB usec: 222181
    URB status: Operation now in progress (-EINPROGRESS) (-115)
    URB length [bytes]: 445
    Data length [bytes]: 445
    [Response in: 60]
    [bInterfaceClass: Unknown (0xffff)]
    Unused Setup Header
    Interval: 0
    Start frame: 0
    Copy of Transfer Flags: 0x00000004
    Number of ISO descriptors: 0
Leftover Capture Data: 4c6f72656d20697073756d20646f6c6f722073697420616d...
````
````
0000   c0 ec 8d 82 9a 9b ff ff 53 03 01 17 02 00 2d 00   Àì....ÿÿS.....-.
0010   d4 7b b1 5e 00 00 00 00 e5 63 03 00 8d ff ff ff   Ô{±^....åc...ÿÿÿ
0020   bd 01 00 00 bd 01 00 00 00 00 00 00 00 00 00 00   ½...½...........
0030   00 00 00 00 00 00 00 00 04 00 00 00 00 00 00 00   ................
0040   4c 6f 72 65 6d 20 69 70 73 75 6d 20 64 6f 6c 6f   Lorem ipsum dolo
0050   72 20 73 69 74 20 61 6d 65 74 2c 20 63 6f 6e 73   r sit amet, cons
0060   65 63 74 65 74 75 72 20 61 64 69 70 69 73 63 69   ectetur adipisci
0070   6e 67 20 65 6c 69 74 2c 20 73 65 64 20 64 6f 20   ng elit, sed do 
0080   65 69 75 73 6d 6f 64 20 74 65 6d 70 6f 72 20 69   eiusmod tempor i
0090   6e 63 69 64 69 64 75 6e 74 20 75 74 20 6c 61 62   ncididunt ut lab
00a0   6f 72 65 20 65 74 20 64 6f 6c 6f 72 65 20 6d 61   ore et dolore ma
00b0   67 6e 61 20 61 6c 69 71 75 61 2e 20 55 74 20 65   gna aliqua. Ut e
00c0   6e 69 6d 20 61 64 20 6d 69 6e 69 6d 20 76 65 6e   nim ad minim ven
00d0   69 61 6d 2c 20 71 75 69 73 20 6e 6f 73 74 72 75   iam, quis nostru
00e0   64 20 65 78 65 72 63 69 74 61 74 69 6f 6e 20 75   d exercitation u
00f0   6c 6c 61 6d 63 6f 20 6c 61 62 6f 72 69 73 20 6e   llamco laboris n
0100   69 73 69 20 75 74 20 61 6c 69 71 75 69 70 20 65   isi ut aliquip e
0110   78 20 65 61 20 63 6f 6d 6d 6f 64 6f 20 63 6f 6e   x ea commodo con
0120   73 65 71 75 61 74 2e 20 44 75 69 73 20 61 75 74   sequat. Duis aut
[... 79 lines cut ...]
    URB usec: 223319
    URB status: Success (0)
    URB length [bytes]: 61
    Data length [bytes]: 61
    [Request in: 38]
    [Time from request: 0.001348000 seconds]
    [bInterfaceClass: Unknown (0xffff)]
    Unused Setup Header
    Interval: 0
    Start frame: 0
    Copy of Transfer Flags: 0x00000204
    Number of ISO descriptors: 0
Leftover Capture Data: 756e7420696e2063756c706120717569206f666669636961...
````
````
0000   40 68 cd 69 9a 9b ff ff 43 03 81 17 02 00 2d 00   @hÍi..ÿÿC.....-.
0010   d4 7b b1 5e 00 00 00 00 57 68 03 00 00 00 00 00   Ô{±^....Wh......
0020   3d 00 00 00 3d 00 00 00 00 00 00 00 00 00 00 00   =...=...........
0030   00 00 00 00 00 00 00 00 04 02 00 00 00 00 00 00   ................
0040   75 6e 74 20 69 6e 20 63 75 6c 70 61 20 71 75 69   unt in culpa qui
0050   20 6f 66 66 69 63 69 61 20 64 65 73 65 72 75 6e    officia deserun
0060   74 20 6d 6f 6c 6c 69 74 20 61 6e 69 6d 20 69 64   t mollit anim id
0070   20 65 73 74 20 6c 61 62 6f 72 75 6d 2e             est laborum.
````
</details>

dmitrystu, 2020-05-05:
BTW
> ```cpp
>std::min(readable.second, size_t(CDC_DATA_SZ))
> ```
I'm not so familiar with C++, but is this correct for the transfer size calculation? AFAIK ```size_t``` is a type, not a function nor a macro function. Can this expression evaluated as 0?

dzarda, 2020-05-05:
`size_t(CDC_DATA_SZ)` is the proper way of casting integral types in C++. Same as `(size_t)value`

dzarda, 2020-05-05:
> Looks like you received a zero-length packet

Interesting, I thought that was the response packet of *this* OUT transaction (that's what WS said)

dmitrystu, 2020-05-09:
This means that ````usbd_ep_write```` was called with zero in the size argument.

dzarda, 2020-05-11:
It's difficult for me to imagine how `if (readable.second > 0) {` evaluates as true 🤔

dzarda, 2020-07-06:
Works flawlessly, the error was im my use of HAL_DMA_PollForTransfer() that reports READY in case of ongoing transfer... Closing

closed 2020-07-06 by dzarda