dmitrystu/libusb_stm32 #54: Remove PMA size warning

dzarda, 2020-05-05:
What's the reason of having the PMA size configurable? Isn't it fixed for a concrete part number?

dmitrystu, 2020-05-05:
It was defined in the include headers for some devices. So, this message is just for information and can be removed or moved to ```#pragma message```

dmitrystu, 2020-05-05:
Also, PMA memory is shared between CAN and USB for some devices. In this case, USB_PMASIZE must be defined manually to avoid data overlapping with CAN.

dzarda, 2020-05-05:
> to avoid data overlapping with CAN.

So you're saying that both CAN be used together? (pun intended)

dmitrystu, 2020-05-05:
I'm not sure about this, but according to section 32.3 of the RM0316 (STM32F303) it looks possible for the F303D/E

dmitrystu, 2020-05-05:
Checked datasheet. Yes, it's possible to use both CAN and USB at the same time. At least for F303xE
````
PA11, PA12 - USB
PB8, PB9 - CAN
````

dzarda, 2020-05-05:
Interesting, F10xxx says

> Note: In low, medium-, high- and XL-density devices the USB and CAN share a dedicated 512-
byte SRAM memory for data transmission and reception, and so they cannot be used
concurrently (the shared SRAM is accessed through CAN and USB exclusively). The USB
and CAN can be used in the same application but /not at the same time.*

dmitrystu, 2020-05-05:
Yeah. Sometimes devices from ST make me crazy. What does it mean _not at the same time_ for the single-core CPU?
OK, It doesn't matter. Will use ```#pragma message``` instead of ```#warning```. It has no side effect with ```-Werror``` option

dzarda, 2020-05-06:
Cool, closing

closed 2020-05-06 by dzarda