dmitrystu/libusb_stm32 #69: Calculating _rxcnt in usbd_stm32l433_devfs.c wrong?

lulle2007200, 2020-10-24:
What is going on here?
https://github.com/dmitrystu/libusb_stm32/blob/master/src/usbd_stm32l433_devfs.c#L245
If endpoint size is greater than 62, we should align it by 32.
Then left shift by 5 bits to get number of blocks in bits 10:14,
then set bit 15 to 1 to indicate block size of 32 bytes.
So should't it be `_rxcnt = 0x8000 & (epsize << 5)`?
Or am i missing something?

dmitrystu, 2020-10-24:
Yes, probably you are right. I need to recheck code, also asm.

dmitrystu, 2020-10-24:
Yes, you are right. It was a typo. But actually result is the same since bitfield COUNTn_RX[0-9] is r/o.

dmitrystu, 2020-10-27:
BTW. According to USB FS specs, EP size greater than 64 bytes is disallowed.

closed 2020-11-13 by dmitrystu

lulle2007200, 2020-11-14:
Thanks for looking into it.
Also im aware that EP greater size 64 bytes is illegal. Makes me wonder why they even have that weird system to allow EP size greater than 62 bytes, but thats offtopic.

dmitrystu, 2020-11-14:
I've been wondered so many times by ST's solutions that I no longer ask myself "why did they do that way?".

7134956, 2020-11-14:
I use 128 to little increase speed. Worked in windows 10.