dmitrystu/libusb_stm32 #76: how to decide if a usb device is connected and configured?

mttbx, 2021-01-04:
I want to use polling method for get message from the usb device. And I can't  just use usbd_ep_read/usbd_ep_write methods because it may corrupt the configure process. How to decide if a usb device is connected and configured?

mttbx, 2021-01-13:
@dmitrystu Can you answer my question? And I find that it may sometimes crush linux kernel, I don't know why, but I'm trying to figure it out.

dzarda, 2021-01-13:
How about using the `cdc_setconf()` callback?

dmitrystu, 2021-01-13:
Sorry, I'm temporary away from keyboard. I'll answer to you ASAP.

mttbx, 2021-01-14:
@dzarda Thank you, I'll try.
I found that if I restart my board while the host communicating with it, the linux kernel will crush. I think this might be a issue of linux kernel. Is it any way to avoid this kind of thing?

```
[   75.601033] Call Trace:
[   75.601034]  <IRQ>
[   75.601038]  queue_work_on+0x3b/0x50
[   75.601042]  acm_write_bulk+0x70/0x80 [cdc_acm]
[   75.601045]  __usb_hcd_giveback_urb+0x7a/0x120
[   75.601047]  usb_giveback_urb_bh+0xa1/0x100
[   75.601050]  tasklet_action_common.isra.0+0x60/0x110
[   75.601052]  tasklet_action+0x22/0x30
[   75.601054]  __do_softirq+0xe1/0x2d6
[   75.601056]  asm_call_irq_on_stack+0x12/0x20
[   75.601057]  </IRQ>
```

dzarda, 2021-01-19:
Wow nice :D Perhaps a VM might avoid the crash?

mttbx, 2021-01-19:
I'm using the newest kernel. VM might help.

closed 2021-01-19 by mttbx