dmitrystu/libusb_stm32 #77: Device does not enumerate when connected via USB HUBd

yaqwsx, 2021-01-18:


dmitrystu, 2021-01-26:
Sorry for the late reply.
Checked with built-in display hub. See no problems.
````
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
        |__ Port 3: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 3: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 3: Dev 3, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 3: Dev 3, If 3, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 6: Dev 5, If 0, Class=Hub, Driver=hub/3p, 480M
            |__ Port 1: Dev 6, If 0, Class=Hub, Driver=hub/3p, 480M
                |__ Port 1: Dev 7, If 0, Class=Mass Storage, Driver=usb-storage, 480M
                |__ Port 2: Dev 8, If 1, Class=CDC Data, Driver=cdc_acm, 12M
                |__ Port 2: Dev 8, If 2, Class=Human Interface Device, Driver=usbhid, 12M
                |__ Port 2: Dev 8, If 0, Class=Communications, Driver=cdc_acm, 12M
````

yaqwsx, 2021-01-28:
Thank you for your quick response. Based on your comment I tried the demo with Nucleo board, where it worked flawlessly. I examined my board more thoroughly and I found the problem - there are also resistors on the data lines to implement a QuickCharge device. They were wired differently compared to my board with STM32F103 (by mistake). Removing those resistors solved the issue - the hub is probably more sensitive compared to the laptop for weir voltage levels on the data lines.

closed 2021-01-28 by yaqwsx