dmitrystu/libusb_stm32 #79: Issues porting to F7

mgomov, 2021-02-08:
We've been making an effort to port the lib to the F723 using HS, but have been running into a few issues. 

Using initialization code primarily pulled from the HAL. Copied over `usbd_stm32f429_otghs.c` as the base driver, since there seems to be a lot of overlap in the HS cores (other than the number of endpoints and some other constants). Similar to https://github.com/dmitrystu/libusb_stm32/issues/9#issuecomment-423875256 , can't get past the bus reset event, and in dmesg enumeration seems to fail after a number of resets:

`[ 9591.952638] usb 3-3.4.1: device descriptor read/all, error -110
[ 9592.128604] usb 3-3.4.1: new high-speed USB device number 82 using xhci_hcd
[ 9597.772548] usb 3-3.4.1: Using ep0 maxpacket: 8
[ 9603.216525] usb 3-3.4.1: device descriptor read/all, error -110
[ 9603.216578] usb 3-3.4-port1: attempt power cycle
[ 9603.896488] usb 3-3.4.1: new high-speed USB device number 83 using xhci_hcd
[ 9609.360465] usb 3-3.4.1: device descriptor read/8, error -110
[ 9614.480412] usb 3-3.4.1: device descriptor read/8, error -110
[ 9614.760386] usb 3-3.4.1: new high-speed USB device number 84 using xhci_hcd
[ 9620.112367] usb 3-3.4.1: device descriptor read/8, error -110
[ 9625.233309] usb 3-3.4.1: device descriptor read/8, error -110
[ 9625.336350] usb 3-3.4-port1: unable to enumerate USB device`

(error -110 is said to be related to power but I'm not sure how much to trust that here, especially considering that the hsusb examples from cubef7 work fine)

Any potential guidance would be greatly appreciated (places to look, things to verify, resources, etc).

dmitrystu, 2021-02-09:
I tested F745VET6 with F446 driver but only in FS mode. See https://github.com/dmitrystu/libusb_stm32/tree/feat/F7

dmitrystu, 2021-02-09:
Do you have any outputs from Wireshark ?