dmitrystu/libusb_stm32 #80: STM32F429 OTG HS: Windows says: Device Descriptor Request Failed

IndianBoy42, 2021-02-09:
I am mostly just using the cdc_loopback example using IRQ, the only different is it is running on thread as part of a FreeRTOS system.

There are 2 problems actually:

sometimes it gets stuck forever at:
usb/src/usbd_stm32f429_otghs.c:121
		_WBC(OTG->GRSTCTL, USB_OTG_GRSTCTL_CSRST);

but most of the time it gets through that, it gets into poll() a couple of times then no more.
meanwhile on my computer Windows says the USB device is not recognized and in Device Manager it says Device Descriptor Request Failed)

Is this a software problem or hardware problem? I dont know much about USB so its hard for me to tell.

IndianBoy42, 2021-02-09:
Just wanted to add: looking at `dmesg` in linux shows:
```
[269416.421657] usb 3-2: new full-speed USB device number 2 using xhci_hcd
[269416.553661] usb 3-2: device descriptor read/64, error -71
[269416.793872] usb 3-2: device descriptor read/64, error -71
[269417.029656] usb 3-2: new full-speed USB device number 3 using xhci_hcd
[269417.161666] usb 3-2: device descriptor read/64, error -71
[269417.401660] usb 3-2: device descriptor read/64, error -71
[269417.513666] usb usb3-port2: attempt power cycle
```

dmitrystu, 2021-02-09:
It must be a hardware problem. Try to comment https://github.com/dmitrystu/libusb_stm32/blob/a9836043879ec60c205644bedf2982548b639ae6/src/usbd_stm32f429_otghs.c#L125 and https://github.com/dmitrystu/libusb_stm32/blob/a9836043879ec60c205644bedf2982548b639ae6/src/usbd_stm32f429_otghs.c#L126 Looks like this is not necessary and will stuck MCU sometimes.

IndianBoy42, 2021-02-09:
Do you have any tips about where the hardware problem is?
I have 33R series resistors on DP and DM, ID is floating and no pull up resistor (stm says there is the embedded pull up on DP)

dmitrystu, 2021-02-09:
I have no tips, but I removed "core soft reset" code from most OTGFS drivers when I faced the same problem on the test. Anyway, the core will be reset automatically when BUS RESET will be issued by the host.

commit d6924e0e (2021-05-02): fix CDC_SET_LINE_CODING

closes #80
closed 2021-05-02 by dmitrystu