dmitrystu/libusb_stm32 #83: Porting to G0x1

dzarda, 2021-03-01:
I reckon the upcoming [G0C1/G0B1](https://www.st.com/resource/en/reference_manual/dm00371828-stm32g0x1-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) ought to be successful. We should add support for their USB cell.

They call it *USB full-speed host/device interface*, and it appears that they extended the existing dev/fs with backwards compatibility. Host bits seem to be a simple addition in register description, occupying the high halfword.

[image]

<details>
  <summary>G4 map for comparison:</summary>
  
[image]  

</details>

One notable change is the 32-bit PMA access:

[image]

-----

Which of our implementation files would be an appropriate starting point?

Cheers

dmitrystu, 2021-03-05:
I think it's close to L052. Unfortunately, there is no proper G0 on the local suppliers stock.

dzarda, 2021-03-05:
Yeah patient we must be

dzarda, 2021-03-12:
Mouser has a few pieces stocked. 
[image]

ghent360, 2023-03-18:
G0s a widely available now, you can get several nucleo G0 kits from mouser.

GrantMTG, 2023-03-22:
That's good news!

ghent360, 2023-03-27:
The PMA on the G0 is 32-bit access only. The rest is more or less the same with some reg names changed.

Created PR #123.