dmitrystu/libusb_stm32 #92: gd32 usbfs support

mttbx, 2021-05-26:


mttbx, 2021-05-26:
It looks like f105 usbfs, but I can't get it worked.

mttbx, 2021-05-26:
Here's the error when I try f105 driver:
```
[  464.507201] usb 1-4: new full-speed USB device number 30 using xhci_hcd
[  469.546739] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[  474.921936] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[  475.129804] usb 1-4: device not accepting address 30, error -62
[  475.129873] usb usb1-port4: unable to enumerate USB device
```

mttbx, 2021-05-26:
I can read message from the host. But I can't send message to the host. I stuck at get_descriptor response. The Urb in the host shows protocol error. Any suggestion would be greatful.

dmitrystu, 2021-05-29:
Just took a test with GD32F103C8T6. It works as STM32F103C8T6.

dmitrystu, 2021-05-29:
What kind of Giga Devices MCU you are trying to implement?

mttbx, 2021-05-31:
@dmitrystu Yes, gd32f103 works fine. I use gd32e103. It supports otg.

mttbx, 2021-05-31:
datasheet: https://www.gigadevice.com/zh-hans/datasheet/gd32e103xxxx-datasheet/
manual: https://www.gigadevice.com/zh-hans/manual/gd32e103xxxx-user-manual/