dmitrystu/libusb_stm32 #95: Get data for USB HID Keyboard's LED via Class Request 0x09 (Set Report)?

GrantMTG, 2021-08-19:
This is a standard feature of HID Keyboards, they receive the LED information (Caps Lock, etc) via a SETUP then a one-byte EP0 OUT data packet.  The setup looks normally like this:

`21 09 00 02 00 00 01 00`

I can set the usbd_process_eprx() function to receive the data (1 byte) using the following, but I don't know the correct way to direct that byte to my own buffer. Alternatively I would need notification or a callback to get it out of the default(?) one. This code currently resides in cdc_control() in the demo and is similar to https://github.com/dmitrystu/libusb_stm32/issues/32#issuecomment-453434318.

```
if((req->bmRequestType & (USB_REQ_TYPE|USB_REQ_RECIPIENT)) == (USB_REQ_CLASS|USB_REQ_INTERFACE))
   {
   switch(req->bRequest)
      {
      case USB_HID_SETREPORT:          // 0x09
           ReportType = req->wValue >> 8;         // 2=Output report, 3=Feature report
           ReportID   = req->wValue &0xFF;
           // Configure the data stage:
           dev->status.control_state = usbd_ctl_rxdata;
           dev->status.data_count = 1;
// Not sure how to get the results from the data stage afterward...
//           dev->status.data_buf = &HidKbLEDs;
//           dev->status.data_ptr = &HidKbLEDs; 
           return usbd_ack;                       // We are expecting an OUT data stage with the info...
```
How can I get the data and/or be notified that is has been received?

GrantMTG, 2021-08-19:
[image]

dmitrystu, 2021-08-20:
Hi
Yes, that's correct.
![Main logic](https://mermaid.ink/img/eyJjb2RlIjoiZ3JhcGggVERcbiBFMChbQ29udHJvbCByZXF1ZXN0IHJlY2VpdmVkXSlcbiBFMVtIYXMgY29udHJvbCBjYWxsYmFja11cbiBFMltQcm9jZXNzaW5nIGNvbnRyb2wgY2FsbGJhY2tdXG4gRTNbUHJvY2Vzc2luZyBzdGFuZGFyZCByZXF1ZXN0XVxuIEU0W1NlbmRpbmcgQUNLIG9yIGRhdGFdXG4gRTVbU2VuZGluZyBOQUtdXG4gRTYoW0VuZF0pXG4gRTAtLT5FMVxuIEUxLS1ZZXMtLT5FMlxuIEUxLS1Oby0tPkUzXG4gRTItLXJldHVybiB1c2JkX2FjayAtLT4gRTRcbiBFMi0tcmV0dXJuIHVzYmRfZmFpbCAtLT4gRTNcbiBFMy0tcmV0dXJuIHVzYmRfYWNrIC0tPiBFNFxuIEUzLS1yZXR1cm4gdXNiZF9mYWlsIC0tPiBFNSBcbiBFNC0tPkU2XG4gRTUtLT5FNlxuICAiLCJtZXJtYWlkIjp7InRoZW1lIjoiZGVmYXVsdCJ9LCJ1cGRhdGVFZGl0b3IiOmZhbHNlLCJhdXRvU3luYyI6dHJ1ZSwidXBkYXRlRGlhZ3JhbSI6ZmFsc2V9)

closed 2021-09-01 by GrantMTG