dmitrystu/libusb_stm32 #105: USB Compliance test failed (Chapter9)

dmitrystu, 2021-12-23:
One thing I have found, that I have not had time to investigate yet (and this is F103), but with polling mode my USB device fails the USB Check compliance suite. With interrupt mode it passes. So maybe look for something unrelated such as that.

I do have a Black Pill board, so at some point I will port my app to that and can test your issue then.

_Originally posted by @GrantMTG in https://github.com/dmitrystu/libusb_stm32/issues/101#issuecomment-997415515_

GrantMTG, 2021-12-23:
I did not find my report (the test saves them but I must have moved it), I probably can't access my hardware now until Jan 3.  I am using an STM32F103.  Eventually I will switch to F4 or G0 maybe.

This library is working well for me (HID "raw", HID KB with multimedia and mouse). Eventually CDC and MIDI too.

dmitrystu, 2021-12-23:
With USB3CV 2.2.2.0 and demo code.
Discovered a problem on TD9.9 "Halt Endpoint Test" with ASM driver. Will be fixed soon. No other failures were found except a small bunch of warnings related to unhandled GetInterface and SetInterface requests.
Tested with F303 and L053.

GrantMTG, 2021-12-23:
OK great, thanks for that. Perhaps my issue was of my own making.

I have another request if I may. Can you try building for USBD_STM32F103? I get a warning about PMA_SIZE not being defined and using a size of 512.  (Keil doesn't support #pragma message so I had to modify it). Is that an issue from an STM header?

dmitrystu, 2021-12-23:
With the latest update see no more problems on F303 F103 L053 L100 for any ASM/C Interrupt/Poll.

dmitrystu, 2021-12-23:
> I have another request if I may. Can you try building for USBD_STM32F103? I get a warning about PMA_SIZE not being defined and using a size of 512. (Keil doesn't support #pragma message so I had to modify it). Is that an issue from an STM header?

See https://github.com/dmitrystu/libusb_stm32/issues/54

closed 2022-02-28 by dmitrystu