dmitrystu/libusb_stm32 #110: cdc_loop example at high speed

EmbeddedMagic, 2022-04-01:
Hi, I am using a blue pill with STM32F103 and the cdc_loop example. To evaluate the stability, I have a PC software that sends an 8byte number to the device. The device will echo the number and the PC will compare if TX=RX. Then the PC increments the number by 1 and send again to the device....all this happens at maximum speed.
It happens that the device is not echoing after 100...3000 transmission. It seems 

```
static int32_t ep_write(uint8_t ep, void *buf, uint16_t blen) {
...
    /* invalid or not ready */
    default:
        return -1;     <-- 
}
```

-1 is returned. What could be the route cause of that?
Thx a lot for your response!

GrantMTG, 2022-04-01:
Maybe you could copy the result to a local variable and see the reason (the value at *reg):

    volatile uint16_t *reg = EPR(ep);
    switch (*reg & (USB_EPTX_STAT | USB_EP_T_FIELD | USB_EP_KIND))
       :

EmbeddedMagic, 2022-04-01:
```
volatile uint16_t *reg = EPR(ep); static uint16_t regCopy ;
....
    /* invalid or not ready */
    default:
        regCopy = *reg;
        return -1;
```
regCopy = **0x3031**
eb=0x81

GrantMTG, 2022-04-01:
Well, FWIW, I never wrote at that level, I always used usbd_ep_write().

I'm going to hazard a guess that the endpoint isn't ready, try again later... you could implement that and see if the situation resolves if you do retry.

It's probably the middle of the night where Dmitry is and/or the war might be affecting him or his locale.