dmitrystu/libusb_stm32 #149: STM32F411, usbd_stm32f429_otgfs.c, USB_CDC_SET_LINE_CODING reads wrong value in usbd_process_eprx()

GrantGeno, 2025-09-03:
Hello again everyone. Hoping someone has more insight than I. This again appears to be a problem with the OTG_FS reading EP0, this time as a result of the host PC issuing SET_LINE_CODING and the STM32 reading garbage (my best guess anyway). This is not so much about the SET_LINE_CODING so much as the general failure to _read_ data stage from Endpoint 0.  (This works OK on the STM32F103 using usbd_stm32f103_devfs.c driver).

Scenario is using the default CDC Loop Demo firmware on an STM32F411E-DISCO. No terminal application running - just enumeration.  

- Initially the host PC (Windows) requests the current Line Coding from the CDC device (odd perhaps, but it does nonetheless). 
- The firmware default values for Line Coding are returned to the host OK.  [Green box]
- **Then the PC sends it back. This is where the problem appears to lie.** This is the write from PC to device over EP0. [Purple box]
- Finally the PC requests it once again. We can see on the USB analyzer the garbage Line Coding data the STM32 picked up. [Red box]

USB trace of above:

[image]

Here is the endpoint read usbd_process_eprx() with the bad data:

[image]

Here is that bad data being returned to our cdc_line variable:

[image]

As far as I can tell, this is the same mechanism that causes the USB HID keyboard LED report to fail as well.  See my previous issue report #148 

I will keep looking into it, but any help is appreciated.

Solartraveler, 2026-09-10:
Okay, since here are some usbd_stm32f429 unsers with issues, maybe you are interested what I found out the last days:

1. Calling usbd_ep_config without calling usbd_ep_deconfig results in a new hardware buffer to be allocated at the end of all already allocated buffers.
So having 1280 bytes of buffer and two endpoints needing 64 bytes on each call, set_tx_fifo fails after ~20 calls.

The cdc_loop.c cdc_setconf() always calls this in the configuring device case. My host OS does this on every device open and never calls the deconfiguring device case. So keeping track if usbd_ep_config has been called and call usbd_ep_deconfig before fixes the issue that a program suddenly can no longer communicate with the device.

2. If USB is able to send data to the device faster than these can be processed (USB mass storage for me), the endpoint needs to go to NAK. But when not reading these data, the USB_OTG_GINTSTS_RXFLVL in OTG->GINTSTS never gets cleared and the while() in evt_poll becomes an endless loop. Other drives like stm32l052_devfs.c don't have a loop in evt_poll at all.
Currently I solved this by adding a return value to the callback telling the loop if it should leave or not.
But maybe removing the loop is a better option?

After solving this, there are two sources of endless ISRs left:

2.1. The not read data still set the RXFLVL flag, so disable this ISR source. (And enable it again once the main loop has processed enough data to actually clear the NAK)

2.2. A SOF ISR enters evt_poll, finds the RXFLVL flag still set and skips clearing the SOF flag.
    So in the loop the RXFLVL flag may only be go into the code for the RXFLVL if the ISR source is not disabled.

With the stm32l052_devfs.c I had no such issue, it looks like the STM32L052 triggers its USB ISR by a rising edge and not by a level like the STM32F405 I am using.

I hope this helps some users of this driver.

GrantGeno, 2026-09-10:
@Solartraveler thank you! I don't have anything meaningful to add (yet), but I will try and come up with some meaningful USB stress tests when I get time.

KimJorgensen, 2026-09-26:
@Solartraveler Regarding the endless loop, it seems to be caused by [this commit](https://github.com/dmitrystu/libusb_stm32/commit/d39665bad43cf3f0e84496833724ccb3abcd8a5b#diff-36c84ec67cd3115cd90c1eb3f1e7f441dc25980d1301bc2ea8dd0203d15da737) and that does not appear to be intentional.

Solartraveler, 2026-10-02:
@KimJorgensen Thanks, yes looks like non intentional. I fixed the loop for me:

https://github.com/Solartraveler/led-toybox/commit/bcc81d8ae18c6869a2f3c5d0ace7709ad5a4c0b8

And added some helper functions to disable the RXFLVL interrupt.

Then I tried to improve the speed in the device -> host direction and had partial success (enough for me). I now can queue up to 7 packets and get a reasonable speed. ChatGPT pointed me to a link that I am not the only one failing to queue in more packages:

https://community.st.com/stm32-mcus-embedded-software-32/another-usb-otg-fs-quirk-the-3-bit-counter-of-in-endpoints-136937

If anyone is interested in my changes:

https://github.com/Solartraveler/led-toybox/commit/f4f4fafc81839eaf0d03cdc0d15b1ed7d6a5566e

I think my speed-up solution would never make it into the master, because it would be bad API design. For my STM32F405 it is now the fastest to do one call of ep_write with 448 bytes, where for the STM32L452 it would be best to call 2x ep_write with 64 bytes each (two commits further I wrote down some MSC performance measurements). I don't have a good solution here, but for me its good enough - and enough USB digging for probably a long time :P

Probably to get the maximum USB speed, it would be needed to fill the USB FIFO by DMA or busy waiting. Hopes this might help someone.